bborncr/nextion #17: Issue uploading to arduino board

tbagginz, 2016-02-12:
Arduino: 1.6.6 (Windows 7), Board: "Arduino Pro or Pro Mini, ATmega328 (5V, 16 MHz)"

C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SoftwareSerial\SoftwareSerial.cpp:375:6: warning: always_inline function might not be inlinable [-Wattributes]

 void SoftwareSerial::setRxIntMsk(bool enable)

      ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SoftwareSerial\SoftwareSerial.cpp:121:6: warning: always_inline function might not be inlinable [-Wattributes]

 void SoftwareSerial::recv()

      ^

C:\Users\Owner\Documents\Arduino\libraries\nextion-master\Nextion.cpp:235:9: error: exponent has no digits

 >>>>>>> 4ecc8e6778707d477938b3f5a74f648bf0781185

         ^

C:\Users\Owner\Documents\Arduino\libraries\nextion-master\Nextion.cpp: In member function 'unsigned int Nextion::getComponentValue(String)':

C:\Users\Owner\Documents\Arduino\libraries\nextion-master\Nextion.cpp:131:25: warning: left shift count >= width of type [enabled by default]

     value = (temp[4] << 24) | (temp[3] << 16) | (temp[2] << 8) | (temp[1]);//Little-endian convertion

                         ^

C:\Users\Owner\Documents\Arduino\libraries\nextion-master\Nextion.cpp:131:43: warning: left shift count >= width of type [enabled by default]

     value = (temp[4] << 24) | (temp[3] << 16) | (temp[2] << 8) | (temp[1]);//Little-endian convertion

                                           ^

C:\Users\Owner\Documents\Arduino\libraries\nextion-master\Nextion.cpp: In member function 'String Nextion::listen(long unsigned int)':

C:\Users\Owner\Documents\Arduino\libraries\nextion-master\Nextion.cpp:226:1: error: expected primary-expression before '<<' token

 <<<<<<< HEAD

 ^

C:\Users\Owner\Documents\Arduino\libraries\nextion-master\Nextion.cpp:226:3: error: expected primary-expression before '<<' token

 <<<<<<< HEAD

   ^

C:\Users\Owner\Documents\Arduino\libraries\nextion-master\Nextion.cpp:226:5: error: expected primary-expression before '<<' token

 <<<<<<< HEAD

     ^

C:\Users\Owner\Documents\Arduino\libraries\nextion-master\Nextion.cpp:226:7: error: expected primary-expression before '<' token

 <<<<<<< HEAD

       ^

C:\Users\Owner\Documents\Arduino\libraries\nextion-master\Nextion.cpp:226:9: error: 'HEAD' was not declared in this scope

 <<<<<<< HEAD

         ^

C:\Users\Owner\Documents\Arduino\libraries\nextion-master\Nextion.cpp:234:3: error: expected primary-expression before '==' token

 =======

   ^

C:\Users\Owner\Documents\Arduino\libraries\nextion-master\Nextion.cpp:234:5: error: expected primary-expression before '==' token

 =======

     ^

C:\Users\Owner\Documents\Arduino\libraries\nextion-master\Nextion.cpp:234:7: error: expected primary-expression before '=' token

 =======

       ^

C:\Users\Owner\Documents\Arduino\libraries\nextion-master\Nextion.cpp:235:1: error: expected primary-expression before '>>' token

 >>>>>>> 4ecc8e6778707d477938b3f5a74f648bf0781185

 ^

C:\Users\Owner\Documents\Arduino\libraries\nextion-master\Nextion.cpp:235:3: error: expected primary-expression before '>>' token

 >>>>>>> 4ecc8e6778707d477938b3f5a74f648bf0781185

   ^

C:\Users\Owner\Documents\Arduino\libraries\nextion-master\Nextion.cpp:235:5: error: expected primary-expression before '>>' token

 >>>>>>> 4ecc8e6778707d477938b3f5a74f648bf0781185

     ^

C:\Users\Owner\Documents\Arduino\libraries\nextion-master\Nextion.cpp:235:7: error: expected primary-expression before '>' token

 >>>>>>> 4ecc8e6778707d477938b3f5a74f648bf0781185

       ^

C:\Users\Owner\Documents\Arduino\libraries\nextion-master\Nextion.cpp:236:3: error: expected ';' before 'String'

   String temp = "";

   ^

C:\Users\Owner\Documents\Arduino\libraries\nextion-master\Nextion.cpp:242:4: error: 'temp' was not declared in this scope

    temp += String(cmd[i], HEX);//add hexadecimal value

    ^

exit status 1
Error compiling.

tbagginz, 2016-02-12:
Easy fix, but I wanted to make you aware of the errors popping up. Thanks for the work on this, it seems much easier to use than the factory stuff

bborncr, 2016-02-12:
Sorry, due to my lack of github knowledge there was some garbage in the Nextion.cpp file and also some of the example folders were messed up.  I have uploaded again and tested. It all seems to be working now.

closed 2016-02-16 by bborncr