bborncr/nextion #18: RS232 commands

yo2ldk, 2016-06-01:
Hi, can I send, only with this display, some digital commands trough rxd txd ?
For example if I make a button ON, when i press him, this send via rs232 
$18	$00 and for OFF button    $18	$01
  thank you in advance !

bborncr, 2016-06-01:
Sorry I don't understand. What exactly is the question relating to our
library?

yo2ldk, 2016-06-01:
Thank you for reply,
  I'm sorry, maybe is not relating with your library, it was more a help hope..
My question it was if is a way to send in user code, only with nextion display,
 without arduino module, some 8 bits data trough rs232 to an equipment,
more exactly , to control a radio station via CAT commands 
  my best regards

bborncr, 2016-06-01:
The Nextion responds to serial commands, you can send and receive the
commands using any device you wish. The Nextion instruction set can be
found here:
http://wiki.iteadstudio.com/Nextion_Instruction_Set

yo2ldk, 2016-07-06:
Hi Team,

  I disturb you again, because I need to find a solution for my work..Thank you in advance for your 
patience and help !
  Now, my board is leonardo mini, and I use from library example "ReceiveMessages" 
My needs is: when i push a button on display who send trough serial char "A"  ( printh 41 ), I want to see it on pc display as "A" via USB Arduino connection
The example look to made what I need, but is nothing on screen when I act the button.
TX from Nextion is connected on Leonardo on Pin 2, RX on pin 3 (2=SCL; 3= SDA)
 I try also on RXD, TXD from arduino (pin 0 and 1) and changed in sketch, but without results
  please some tips