bborncr/nextion #19: Read from Nextion Screen

baarini, 2016-07-02:
Hi,

I noticed that these Libraries doesn't have the features to read inputs from Nextion to Arduino.
i.e. read the Text Component Value to Arduino or the Number Component value to Arduino.
Does anyone has such a code to do it?

Thanks.

Operrr, 2016-12-08:
int i=myNextion.getComponentValue("page0.b0");  - Worked
string i=myNextion.getText("page0.va0"); - No worked

AndreasKouv, 2019-01-25:
> string i=myNextion.getText("page0.va0"); - No worked

you try to get text from Nextion, but as I understand you have a "numeric" variable which is not text.
try to convert the "text_data" you have received into "int" type data.

baarini, 2019-01-26:
Thanks, I'll try it!

raulcq01, 2019-02-14:
i need to save a text value to char from Text component in arduino, any ideas?