bborncr/nextion #33: not at all as expected in editor

RootlessAgrarian, 2018-01-28:
Hi, I'm a complete Nextion Noob trying this tutorial for the first time.  I figured my recent Nextion probably was incompatible with the old tft files so I imported the HMI file for my rez into the Nextion Editor:  272x480Example.HMI

What I got was unexpected.  The basic image seems to be reversed L/R and there is only 1 page.  I expected 3 pages and the image to be right way around.  Here's a link to screen snapshot

[img]https://i.imgur.com/5zYxpQ2l.png[/img]

So ummm... is this tute now so out of date that there is no hope for it & I must start from scratch, or is there some simple hack I can do to make the HMI file work...?

hbadotti, 2018-01-29:
Go to "Device", then Display and try do change from 0 to 180 or 90 to 270.
[image]

RootlessAgrarian, 2018-01-30:
Hi bborncr

thanks for the suggestion, but the image is still reversed r/l no matter
what rotation it is in.  I tried all possible rotations (the landscape ones
of course clip the image) and it remains reversed in all of them.

it is possible that ITEAD changed the conventions for the HMI file since
you created your demo...  I am proceeding with my own bkgnd image, starting
from scratch;  it takes a little longer than just loading your demo, but
it's educational.

I do like your lightweight nextion library :-)  compared to ITEAD's very
heavy, large library, it seems more in the Arduino spirit.

hbadotti, 2018-01-30:
When you upload the file to your Display, it keeps reversed?