bborncr/nextion #44: Weired output on getComponentValue

schultzgc, 2019-11-05:
Hi there,

hopefully this lib is still supported by the community.
I´ve got some issues with the getComponentValue function.
I´m using the following code on my ESP32 using Hardware Serial2.
As soon as I toogle my button it should output the toogle button val on my serial monitor,
but I get this weired output. What did I wrong?

`
65 0 2 1 ff ff ff
Toogle Button
-1
q???
`

`
#include <Nextion.h>

Nextion myNextion(nextion, 9600);

void NextionInit() {
  if (myNextion.init()) {
    Serial.println("Nextion Ready...");
  } else {
    Serial.println("Nextion not responding...");
  }
}

void NextionListen() {
  String message = myNextion.listen();
  if (message != "") {
    Serial.println(message);
    if (message == "65 0 2 1 ff ff ff") {
      Serial.println("Toogle Button");
      int value = myNextion.getComponentValue("bt0");
      Serial.println(value);
    } else if (message == "65 0 3 1 ff ff ff") {
      Serial.println("Simple Button");
      myNextion.setComponentValue("bt0", 1);
    }
  }
}
`

schultzgc, 2019-11-06:
Changing the timeout from 20 to 200 in Nextion.cpp did the trick.

closed 2019-11-06 by schultzgc

hayderismael, 2022-10-16:
> Changing the timeout from 20 to 200 in Nextion.cpp did the trick.

That answer saved me. I always get a wrong return of my inputs on Nextion, and I went to other slow libraries. Now, after changing the timeout, all is perfect.